floodnet-nyc/flood-sensor #19: Remote app change for deployed devices

cmydlarz, 2021-06-01:
Each device is setup with a default appkey pointing to our dev application. On first boot, device receives onJoin downlink containing deployment apppkey. We catch this first onJoin event and inject the new appKey. The appkey to send is decided by deveui. We maintain database of devEUIs to AppKeys.

jatinpalchuri, 2021-06-07:
So as soon as we receive a  downlink we have to update  the key values  and  lmic library should be reset thereby making the mcu point to a different app.([code]( https://github.com/floodsense/floodsense_sensor/blob/v3-Appkey-change/src/lorawan.cpp))
While implementing this there is an error assertion  in lmic library here [Assertion error ]( https://github.com/mcci-catena/arduino-lmic/blob/48f04f1413a7f10bf52d979fe01c4d00911100d8/src/lmic/radio.c#L1065)
Is there a better way to stop or check for completion of all tasks in lmic library?

jatinpalchuri, 2021-07-19:
The downlink is not always received by the micro controller  we are facing a similar issue to  [this](https://github.com/TheThingsNetwork/lorawan-stack/issues/1228)

Praneethsvch, 2021-11-16:
bac3104c23c4484a1277ef45b0ecec923dfcc22a Latest remote app change commit - Verified

closed 2021-11-16 by Praneethsvch