floodnet-nyc/flood-sensor #61: RCA - Russell St 2

ak6492, 2023-06-09:
**Sensor Deployed:**
- March 5, 2021 by FloodNet

**Sensor Version:** 
- v2.0.

**Firmware Version:**

**Sensor Down:**
- April 3, 2023 19:23

**Solution Implemented:** 
- whole sensor replaced on June 7, 2023 with another v2.0.

**In Lab RCA:** 
- antenna was perfectly intact 
- no visible rusting or corrosion both internally and externally
- battery in sensor completely drained and dead 
- solar panel outputting voltage and increases it's output voltage when light is shined on it thus proving it is functioning correctly 
- sensor booted back up when new battery was inserted into it  

**deployment_id:** easily_dear_mouse

**dev_id:** fs-00023 

**Fixes/Action:**
In conclusion, sensor appears to have gone down due to the battery dying completely and being unable to retain charge from solar panel. The reason behind this seems to be the battery just being out in the field for about two years and going through its natural battery life cycle. Need to track about how long after being out in the field, batteries stop being functional to try to get ahead of sensors going down due to faulty old batteries.

mprafull1, 2023-06-09:
In Lab RCA:
Battery didn't got charged even when connected to the charger.

ak6492, 2023-07-10:
Sensor replaced on 06/07, went down again on 06/09, data analysis shows the following: 

[image]

[image]

[image]

Signal Strength looks good, battery is trending down rapidly and not charging

ak6492, 2023-07-12:
**07/11 On Site Analysis:** 

battery was visibly puffed up. ended up replacing battery and solar panel.

mprafull1, 2023-07-13:
Battery was dead also but it charged in the lab.

ak6492, 2023-11-10:
Sensor data has repeatedly been noisy. On site visit on 11/09 did not reveal any visible sign of what might be causing the noise. Nothing was obstructing the sensor and there was no grass or spiderwebs inside the pipe. Need to observe and understand this issue further. 

[image]

ak6492, 2024-02-16:
sensor removed on 02/15

closed 2024-02-16 by ak6492