floodnet-nyc/flood-sensor #65: Gateway - 303 Vernon

ak6492, 2023-06-30:


mprafull1, 2023-07-12:
[image]
[image]

SNR and RSSI value of additional sensor (fs-00085) and glamos at Knickerbocker Ave/Hart St

mprafull1, 2023-07-12:
No RSSI and SNR at Morgan Ave/Johnson Ave as they were not joining at all.

mprafull1, 2023-07-12:
SNR and RSSI at Vernon 303

[image]
[image]

At 0.2 mile from gateway

[image]
[image]

At 0.4 mile from gateway

[image]
[image]

At 0.6 mile from gateway

[image]
[image]

At 0.8 mile from gateway

[image]