floodnet-nyc/flood-sensor #75: Sensors changing modes

mprafull1, 2023-08-03:
Sensors at the locations mentioned below are going offline on the dashboard but are sending data on the things network.

Sensors are going into the conflicting mode again and again.  It is required to send the start data packet to these sensors for them to show on the dashboard but again after some time they go into conflicting mode.
fs-00020 ( Lorraine & Henry)
fs-00019 ( Dwight and King)
fs-00002 ( 4th Av/Garfield Pl)

cmydlarz, 2023-11-07:
Can we drop the FW version here so we can solve this bug? Its being seen a lot and leads to a large amount of data loss @ak6492 @Praneethsvch @mprafull1

Praneethsvch, 2023-12-14:
linked to [3207409](https://github.com/floodnet-nyc/flood-sensor/commit/32074090ad07cdc7a972c6ce96447420c3d4c40c)

closed 2023-12-14 by Praneethsvch