floodnet-nyc/flood-sensor #80: RCA - BK: Wallabout/ Throop Ave

ak6492, 2023-09-01:
Sensor Deployed:
March 9, 2023 by FloodNet

Sensor Version: 2.5.

Firmware Version:

deployment_id: mean_flying_fish

Sensor has been very noisy since deployment. Upon inspecting the sensor on 08/31 it was seen that the sensor was very crooked and not leveled at all on the drive rail. The antenna was also damaged, and looked barely operational. The sensor was remounted to be more leveled on 08/31. Need to see if this will fix the noise issue for this particular sensor 

[image]
[image]
[image]

cmydlarz, 2023-09-05:
Sensor is being observed before we close the issue. New label assigned for possible fix

ak6492, 2023-09-08:
Observed on 09/08, seems to no longer be getting no returns but may still be noisy

ak6492, 2023-09-15:
Observed on grafana on 09/15, seems to be doing a lot better. At most one no return value a day. Closing issue for now

closed 2023-09-15 by ak6492