floodnet-nyc/flood-sensor #86: RCA - BK: Brinkerhoff Ave / 157th St

ak6492, 2023-09-22:
Sensor Deployed:
September 27, 2022 by FloodNet

Sensor Version: 2.0.

Firmware Version:

Sensor has been hitting no return values pretty much since deployment. 

[image]

ak6492, 2023-09-22:
09/21: This sensor is known to be an illegal dumping ground. Would recommend moving the sensor to another location.

mprafull1, 2023-09-22:
[image]

ak6492, 2024-04-01:
03/28: somehow more trash present now. very noisy and consistently shows up wrong data on dataviz. strongly recommend replacing it 

[image]