floodnet-nyc/flood-sensor #90: RCA - Q: Kissena Blvd/ Rose Ave

ak6492, 2023-12-12:
Sensor Deployed:
November 16, 2023 by FloodNet

Sensor Version: v3.1. (cell)

Firmware Version:

Sensor Down:

deployment_id: doubtfully_popular_fawn

Sensor never charged since it had been deployed. Went down on November 25, 2023

ak6492, 2023-12-12:
12/12: In lab RCA that the solar IC had not been soldered properly. The solar IC was incapable of charging. Replaced sensor with new one.

ak6492, 2023-12-12:
follow up and check if sensor is charging correctly since replacement in a week

Praneethsvch, 2024-01-16:
### Issue
- Sensor failed due to water ingress
- When water creeps in the sensor still operated but in bad conditions 
    - noisy data like battery data below from water damaged sensor 
    - It starts trending down or doesn't charge as well as the other sensors 
  
[image]

  ### Metadata
  1. `doubtfully_popular_fawn `
  2. `fsc-00008` is damaged and now replaced with `fsc-00015`

  ### Fixes/Action
  - swapped with a new sensor 
  - To swap the plug with the n-type antenna ASAP

ak6492, 2024-02-12:
02/12: Down completely since plug was swapped out with n-type antenna on 01/24. Did struggle to connect via nodered on the ground

ak6492, 2024-03-05:
02/28: the switch was turned off physically. no other visible damage. up and running as of 03/05

closed 2024-03-05 by ak6492