floodnet-nyc/flood-sensor #102: RCA: Q - 88th Street Alleyway/Astoria Blvd

ak6492, 2024-03-05:
Sensor Deployed:
January 12, 2024 by FloodNet

Sensor Version: v3.1. (cell)

Firmware Version: 

Sensor Down: February 29, 2024

deployment_id: curly_furry_squirrel

down unclear yet as to why. Possible sources of failure:
1. connectivity issue
2. battery connection 
3. water damage 
4. physical damage to the sensor

ak6492, 2024-03-18:
suspected cause is sensor has been damaged or shut down

ak6492, 2024-03-22:
03/22/24: sensor was visited on March 21, 2024. Simple SIM card replacement fixed the issue. Suspect that the SIM card had been expired. Double check and confirm

cmydlarz, 2024-03-22:
Both SIMs that came back were working in the lab:

ak6492, 2024-03-25:
@cmydlarz we did reboot and checked battery level before swapping